CANopenNode/CANopenEditor #2: Compile errors from OD.h and OD.c due to PDO modification

woudie, 2021-07-12:
Hi,

I've been trying to re-map some PDOs using CANopenEdtior, but for some reason, the generated OD.h and OD.c seem to fail when trying to compile with the rest of my source code. After some trial and error, I figured that this occurs only after editing a PDO. The error log is shown in the image. Any advice or assistance on resolving this issue would be greatly appreciated!

[image]

woudie, 2021-07-12:
Additionally, it seems that the missing attributes were all SDO permissions

woudie, 2021-07-12:
Newer update: Most of this problem seems to occur when I try using the TPDO or RPDO tab, but if I use the object dictionary tab to edit the PDOs, it seems mostly fine. I say mostly because the counters of OD objects in the OD.h were still screwed up, but at least I was able to modify the PDOs.

nkonidaris, 2021-10-02:
I'm facing the same problem, is there any update on a fix for this or have I misconfigured the PDOs ?
[image]

woudie, 2021-10-03:
From what I remember, the RPDO and TPDO mapping tabs are completely broken. By that I mean, the generated OD after modifying anything on those tabs will break your OD even if you enter everything correctly. As long as you only use the Object Dictionary tab, your generated OD will be fine. Hopefully somebody comes along and fixes those tabs haha

andrew-elder, 2022-04-28:
Has this been fixed in mainline code yet?

Regelink, 2022-06-01:
No, pull request #1 needs to be merged by the maintainer. In the mean time, feel free to use my fork [Regelink/CANopenEditor](https://github.com/Regelink/CANopenEditor)

trojanobelix, 2022-08-31:
Can someone provide an xdd to reproduce this behavior?

CANopenNode, 2022-08-31:
I apologize for my inactivity here. I currently can't find my time to be more active. (But at least I receive notifications from now on.)

@trojanobelix is now co-maintainer. Thank you for your recent commits to the repo.

trojanobelix, 2022-09-17:
TPDO or RPDO tab should be fixed in the bugfix branch. Please test and create a new issue if still something wrong.

closed 2022-09-17 by trojanobelix