CANopenNode/CANopenEditor #8: Issues with RX PDO Mapping

johndcollins, 2021-08-12:
I have a EDS with 40 writeable analog outputs. I'm trying to map these on the RX PDO Mapping screen, but running into issues.

It seems the Getbase(string defaultvalue) function is returning the wrong fromBase number for different values.
For example 0c will return base 10 instead of 16 and 08 will return base 8 instead of 10.
So it's impossible to map these outputs to the PDO.

johndcollins, 2021-08-12:
I tried to make a quick hotfix which seems to have fixed the error and proven that this is the issue :
[image]

Also, I discovered another issue. If I drag an item to the last position, it gets placed in the slot before. So if I drag it to byte 6-7 in my case it switches place with byte 4-5 Even if I remove byte 6-7, it removed byte 4-5 instead.

johndcollins, 2021-08-13:
Forgot I also changed this:
[image]

trojanobelix, 2022-09-02:
It should be fixed now in the new "bugfix" branch. Also the drag and drop issue. But the pattern for octal is set to  "@^0[0-7]+" 
 not "@^0[0-7]*$"  so values with e.g. whitespace at the end it still matches.

It would be helpful if you could test the fix.

closed 2022-09-07 by trojanobelix