CANopenNode/CANopenEditor #15: Adding new PDO causes exception!

sudoDavi, 2021-11-19:
Whenever I press the Add new PDO in the TX PDO Mapping tab I get an Unhandled exception!
This error occurs with any file, be it a new EMPTY project or when editing one of the example ones (demoDevice from CANopenNode or the DS301_profile).

[image]

I'm using the latest Windows build

Here is the details from the exception:
```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentNullException: Value cannot be null.
Parameter name: value
   at System.Windows.Forms.ComboBox.ObjectCollection.IndexOf(Object value)
   at ODEditor.DeviceODView.PopulateObjectLists(EDSsharp eds_target) in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceODView.cs:line 197
   at ODEditor.DeviceView.dispatch_updateOD() in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceView.cs:line 92
   at ODEditor.MyTabUserControl.doUpdateOD() in Z:\CANopen\CANopenEditor\EDSEditorGUI\MyTabUserControl.cs:line 62
   at ODEditor.DevicePDOView2.addnewPDO() in Z:\CANopen\CANopenEditor\EDSEditorGUI\DevicePDOView2.cs:line 719
   at ODEditor.DevicePDOView2.button_addPDO_Click(Object sender, EventArgs e) in Z:\CANopen\CANopenEditor\EDSEditorGUI\DevicePDOView2.cs:line 709
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4420.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
EDSEditor
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Users/WDavi/Desktop/Repos/CANopenEditor-build/CANopenEditor-build/EDSEditor.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4400.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4360.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
libEDSsharp
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Users/WDavi/Desktop/Repos/CANopenEditor-build/CANopenEditor-build/libEDSsharp.DLL
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
SourceGrid
    Assembly Version: 4.40.4580.29115
    Win32 Version: 4.40.4580.29115
    CodeBase: file:///C:/Users/WDavi/Desktop/Repos/CANopenEditor-build/CANopenEditor-build/SourceGrid.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
```

CarmenMFM, 2022-02-10:
Hi @sudoDavi !

I have the same problem and i need to add new TPDO to my project. 
When i do a new TPDO in windows I have the same error but when i do it in Linux EDSeditor closes.
How can i do it?

Thanks!

aerospacesoftware, 2022-03-07:
I think I have a fix for this.  Sorry I just figured this was the latest fork of Robin Cornelius' repo.  I can create a pull request if preferred.

diff --git a/libEDSsharp/PDOHelper.cs b/libEDSsharp/PDOHelper.cs
index ea993ba..c66ed87 100644
--- a/libEDSsharp/PDOHelper.cs
+++ b/libEDSsharp/PDOHelper.cs
@@ -408,7 +408,7 @@ namespace libEDSsharp
         public void addPDOslot(UInt16 configindex)
         {
             //quick range check, it must be a config index for an RXPDO or a TXPDO
-            if( (configindex<0x1400) || (configindex >= 0x1a00)  || ((configindex>=1600) && (configindex<0x1800)))
+            if( (configindex<0x1400) || (configindex >= 0x1a00)  || ((configindex>=0x1600) && (configindex<0x1800)))
                 return;

             foreach(PDOSlot slot in pdoslots)
@@ -422,7 +422,10 @@ namespace libEDSsharp
             PDOSlot newslot = new PDOSlot();
             newslot.ConfigurationIndex = configindex;

-            newslot.COB = 0x180; //Fixme need a better default???
+
+            newslot.COB = 0x180;        // Fixme need better defaults???
+            newslot.configloc = "RAM";
+            newslot.mappingloc = "RAM";

             pdoslots.Add(newslot);

trojanobelix, 2022-08-30:
Should be fixed with [4851453](https://github.com/CANopenNode/CANopenEditor/commit/48514531f83847606986d6e2acfb59f51b821651)

closed 2022-09-07 by trojanobelix