CANopenNode/CANopenEditor #23: Editor created OD.c does not compile

xenpac, 2022-06-03:
when using the standard Canopennode example DS301_profile.eds file with the editor and export canopen OD.h/c, i get compile errors.

This also happened when modifying things in my project.

So, editor not usable!

HamedJafarzadeh, 2022-06-03:
Hey,

You might want to give my forked version a try. It is working fine for me.

[https://github.com/HamedJafarzadeh/CANopenEditor](https://github.com/HamedJafarzadeh/CANopenEditor)

xenpac, 2022-06-04:
hi, almost.
seems that the   Counters of OD objects on top of od.h are partly missing.  (doing a fil compare)

xenpac, 2022-06-04:
[image]

xenpac, 2022-06-05:
I solve my own issue:

This canopen editor only works correct, if you use the .xpd  config file format.
with .xdd, .eds, .dcf,  it will NOT work !!!

trojanobelix, 2022-09-20:
Can you please run a test and provide me with the erroneous od.c output and a working file for a file comparison? And of cause the xdd

closed 2022-11-03 by trojanobelix