CANopenNode/CANopenEditor #29: ArrayOutOfBoundsException on loading xdd file

Daimonion1980, 2022-06-27:
I'm wondering why there is no issue about this exception on loading a xdd file into the editor:
[image]

the line is already marked as #WTF ;)

[image]

Changing the subindex to zero get's the editor to work, but don't get the proof that everything else works correct.

Tested with generated xdd files from Kickdrive and Systec tools.

trojanobelix, 2022-09-06:
Can you provide a XXD that reproduces this?

Daimonion1980, 2022-09-15:
Sure.

[objectDictionary_broken.xdd.txt](https://github.com/CANopenNode/CANopenEditor/files/9575527/objectDictionary_broken.xdd.txt)

trojanobelix, 2022-09-19:
should be fixed in 973eeb685557841ad506539557143c8934d5c127. Could you please test and comment. Thanks.