CANopenNode/CANopenEditor #33: Don't know how to compile

JorgePerC, 2022-08-18:
Hello there, I would like to create an ODS from an EDS. 
However, I'm unable to compile the EDSEditor.sln with msbuild (I'm not really good in Linux), nor make

Could you please help me? 😞 

The errors I get are something like this: (are you missing a using directive or an assembly reference?)
Btw, I'm on Ubuntu 22

Thanks in advance

ciandonovan, 2022-08-21:
AFAIK this is a Windows-only application.

However, I have successfully gotten it to run with [Bottles](https://docs.usebottles.com/getting-started/installation), a WINE wrapper/manager

ciandonovan, 2022-08-21:
# CANopenEditor on Ubuntu 22.04

## Install prerequisite packages

`sudo apt update && sudo apt install -y git build-essential mono-runtime libmono-system-windows-forms4.0-cil unzip`

## Fetch release build

`wget https://github.com/CANopenNode/CANopenEditor/archive/refs/heads/build.zip` 

## Unzip and enter archive

`unzip CANopenEditor-build.zip`
`cd CANopenEditor-build`

## Run CANopenEditor

`mono EDSEditor.exe`

JorgePerC, 2022-08-21:
Thanks!!!!
[image]

closed 2022-08-21 by JorgePerC