CANopenNode/CANopenEditor #34: Exporter writes lowercase "i" to "ı" when exporting CanOpenNode

ihsanliturk, 2022-08-23:
I'm using Turkish language settings. When I try to export CanOpenNode command, it exports "identitiy" as "ıdentity". I can correct this with a simple find/replace, but it will be better corrected in the source code.

Here is the output in OD.h file: 

    struct {
        uint8_t highestSub_indexSupported;
        uint32_t vendor_ID;
        uint32_t productCode;
        uint32_t revisionNumber;
        uint32_t serialNumber;
    } x1018_Ä±dentity;

trojanobelix, 2022-09-20:
might be fixed in ea8c040021f9dbbaeaaa311c428bb02a42605d00. Please test!

myildirim10, 2022-12-05:
I was also having problems with Turkish characters. Namely,
- 'ı' instead of 'i'
- 'İ' instead of 'I'
- Some ".attribute = ," content being empty
- Disappearance of "define" symbols in "OD.h"

So I wrote a simple 'python' code. If you run it where your OD.c and OD.h files are, it will make the necessary changes.
You can make minor changes to the python code if there are different bugs that need to be changed.
[Bugfix_OD_Files.zip](https://github.com/CANopenNode/CANopenEditor/files/10151217/Bugfix_OD_Files.zip)

myildirim10, 2022-12-05:
I am also having problems with Turkish characters. Namely,

- 'ı' instead of 'i'
- 'İ' instead of 'I'
- Some ".attribute = ," content being empty
- Disappearance of "define" symbols in "OD.h"

So I wrote a simple 'python' code. If you run it where your OD.c and OD.h files are, it will make the necessary changes.
You can make minor changes to the python code if there are different bugs that need to be changed.
[Bugfix_OD_Files.zip](https://github.com/CANopenNode/CANopenEditor/files/10151244/Bugfix_OD_Files.zip)

trojanobelix, 2022-12-05:
Which version do you have tested?

myildirim10, 2022-12-05:
> Which version do you have tested?

I tested in this version: v4.0-51-g2d9b1ad

trojanobelix, 2022-12-05:
It makes more sense to test the actual binaries where a lot of bug are fixed.
Also this bug (hopefully).

Please give a feedback after testing the actual binaries, if your are happy with the fixes for this problem.

trojanobelix, 2022-12-30:
Will be closed in a few days due to lack of feedback

closed 2023-01-03 by trojanobelix