CANopenNode/CANopenEditor #35: Export documentation broken on linux/mono

aquaid3, 2022-10-02:
When running the latest build v4.0-51-g2d9b1ad exe file on Linux (Lubuntu 20.04 or Debian 11) with mono 6.8.0.105, I get the following error when trying to export Documentation (Ctrl-P), even with a new default OD file:

```
System.ComponentModel.Win32Exception (0x80004005): Cannot find the specified file
  at System.Diagnostics.Process.StartWithShellExecuteEx (System.Diagnostics.ProcessStartInfo startInfo) [0x00129] in <a85c1a570f9a4f9f9c3d2cfa5504e34f>:0 
  at System.Diagnostics.Process.Start () [0x00038] in <a85c1a570f9a4f9f9c3d2cfa5504e34f>:0 
  at (wrapper remoting-invoke-with-check) System.Diagnostics.Process.Start()
  at System.Diagnostics.Process.Start (System.Diagnostics.ProcessStartInfo startInfo) [0x0001e] in <a85c1a570f9a4f9f9c3d2cfa5504e34f>:0 
  at System.Diagnostics.Process.Start (System.String fileName) [0x00006] in <a85c1a570f9a4f9f9c3d2cfa5504e34f>:0 
  at ODEditor.ODEditor_MainForm.documentationToolStripMenuItem_Click (System.Object sender, System.EventArgs e) [0x00111] in <430ff197535c448e93032a5323a38e45>:0 
```
This used to work for me on Linux a while back.  I don't know if it was a change to CANopenEditor or an update to mono that broke things.  Thanks for any help.

trojanobelix, 2022-10-11:
Please test the latest commit of the bugfix branch. It might be solved there. I have not the environment to check it.

aquaid3, 2022-10-11:
Unfortunately, building on Linux also seems to be broken.  (I tried xbuild and dotnet msbuild).

Is there a pre-built .exe for the bugfix branch I can test?

trojanobelix, 2022-10-12:
What does "seems to be broken" mean? Only on the bugfix branch?

aquaid3, 2022-10-22:
Sorry,  by "seems to be broken" I meant that I couldn't get the `bugfix` branch code to build on my linux system.  I also tried building the `main` branch and it throws similar errors.  I don't use mono for anything else, so I'm not confident that mono or dotnet is installed properly or with the appropriate versions.

In case it's helpful to someone, see [linux-compile-bugfix.txt](https://github.com/CANopenNode/CANopenEditor/files/9844645/linux-compile-bugfix.txt) for output from failed `xbuild` and `dotnet build` calls.

But since then, I also realized the newer CANopenEditor versions have an `Export->Documentation (*.md)` option, which does work.  (My original error report above is for `Reports->Documentation` which I think also tries to open a browser to view the newly generated .md file.)  

I can also run the pre-compiled binaries on linux without having to compile anything, so the obvious workaround is to do that and then stay away from the `Reports->Documentation` menu option.

trojanobelix, 2022-10-22:
seems that the .NET Framework Developer Packs are not installed!?

closed 2023-03-09 by trojanobelix