CANopenNode/CANopenEditor #40: Wrong naming for records and array

xGreenPandax, 2022-11-18:
With new release v4.0-102-g634dee9 i have a problem with "ODT_RECORD" and "ODT_ARRAY", because in CANopenNode/301/CO_ODinterface.c it's called "ODT_REC" and "ODT_ARR"

trojanobelix, 2022-11-18:
Uuups. Using V4? Can you put some context lines lines of the created c file (OD.c/h or CO_OD.c/h)

trojanobelix, 2022-11-18:
Got it, please try the latest bugfix branch (634dee9f6c0cd11f68106936843153c896b73711)

xGreenPandax, 2022-11-20:
Sorry for the delay. It looks correct. Could you provide binaries?

trojanobelix, 2022-11-20:
Yes, but it might take some days.

trojanobelix, 2022-11-23:
Released v4.0-103-g029e57e

CANopenNode, 2022-11-29:
It is still not fixed.
I think, the line 172 in https://github.com/CANopenNode/CANopenEditor/commit/029e57e273771e8e8d667e2cb1ade25f39ed30c4 should be:
`odObjectType = odObjectType.Substring(0, 3);`

(Source of the problem: https://github.com/robincornelius/libedssharp/commit/af4d323768b8e065ba6cd7b022a7561671cf507d#diff-d444f21db6a5f27fef9ee46f265a188fc674e24a1502b1150256ac81bf2fa2fb)

I did a quick check. Otherwise exporter to V4 seems to produce correct output.

Could you provide new binaries?

trojanobelix, 2022-11-29:
Janez, you are absolutely right. Sorry, I must have deleted it by mistake. I will provide new binaries!

trojanobelix, 2022-11-29:
Done, please test.

CANopenNode, 2022-11-29:
It works, thanks.

closed 2022-11-29 by CANopenNode