CANopenNode/CANopenEditor #41: More than 3 TPDOs

PascalGuthof, 2022-12-01:
I have another problem/question. But this is better placed in a new issue. 
When I try to set up more than 3 TPDOs, it works in the editor. But the device makes an error when it tries to send this last PDO. As EMCY message comes "CO_EMC_CAN_OVERRUN" together with "CO_EM_EMERGENCY_BUFFER_FULL". 
As soon as I delete the TPDO, the configuration works as planned. 

If anyone knows advice, I have uploaded the xdd files and the OD files.

3PDOs.xdd: https://filehorst.de/d/eazcilwe
4PDOs.xdd: https://filehorst.de/d/eiFsCFgu
OD_3PDOs.c: https://filehorst.de/d/efxqekDa
OD_3PDOs.h: https://filehorst.de/d/eBJdufbz
OD_4PDOs.c: https://filehorst.de/d/enkgGsiJ
OD_4PDOs.h: https://filehorst.de/d/eevfnJEb

CANopenNode, 2022-12-01:
It seems, object dictionary is correct. There must be something else with the device. Can you provide some CAN messages?

PascalGuthof, 2022-12-02:
Thanks for the reply.
Yes I actually also think that the OD is correct. But the successful running of the program depends on that alone. 
Never mind. I have recorded a trace. The 4 PDOs have the ID 1A0, 2A0, 3A0 and 4A0. PDO 1-3 send numbers, PDO 4 is 0. 
The last message is the mentioned EMCY message.
[4PDOs.txt](https://github.com/CANopenNode/CANopenEditor/files/10140504/4PDOs.txt)

I hope this is helpful.

trojanobelix, 2022-12-02:
[4PDOsnew.xdd.txt](https://github.com/CANopenNode/CANopenEditor/files/10141139/4PDOsnew.xdd.txt)
Can you check this one, ignoring the warnings.

CANopenNode, 2022-12-02:
This is certainly a strange behavior. There may be something wrong inside the driver. I think, this is not CANopenEditor issue.

1. Fourth byte inside the emergency message is 0x14.
2. inside CO_emergency.h: `CO_EM_CAN_TX_OVERFLOW = 0x14U,`
3. follow `CO_EM_CAN_TX_OVERFLOW` inside CO_emergency.h, where errors from driver are verified.
4. then follow `CO_CAN_ERRTX_OVERFLOW` inside the driver ...

Otherwise, wrongly configured object dictionary should not trigger CAN TX overflow.

CANopenNode, 2022-12-06:
Currently there is not enough information about the bug. Is it inside a driver? Custom driver or which?

CANopenNode, 2023-03-28:
Must be something with the driver. Please ask there.