CANopenNode/CANopenEditor #43: TDO cannot be removed

franckinux, 2022-12-08:
I use the version 4.0.103.
Create a new project.
Go to "TX PDO mapping" tab.
Click the button "Add new PDO"Add a new TPDO,.
Click the TPDO line to select it.
Click the button "Delete PDO" and confirm the action.
The TPDO is not removed.
Same thing for RPDO.
Thanks.

trojanobelix, 2022-12-08:
Works as expected with 4.0-104-g6f50f73. Please try this release and close if success.

franckinux, 2022-12-08:
Thanks for redelivering version 4.0.104.
It effectively works on Windows, but not on Linux.
I'm running Linux Mint 21.0 with mono.
The workaround is to remove TPDO and RPDO objects in the "Object Dictionary" tab by selecting them and delete them with a right click.

trojanobelix, 2022-12-08:
Bad news. Bur unfortunately I am not able to debug on a Linux system.

CANopenNode, 2022-12-09:
Deleting PDO doesn't work on Linux in my case too, otherwise CANopenEditor works on Ubuntu 22.04 without problems.

Anyway, I think, CANopenEditor currently works good enough. Code is quite complex. To make it more robust, some parts should be rewritten and made more systematic, including the base (that would take huge amount of time). Original author is currently not active.

Of course fixes are welcome, but with direct editing of the object dictionary workarounds are possible.

franckinux, 2022-12-09:
As there is a workaround (I didn't know that before wrtiing this issue), the situation is good enough. It can be closed.

CANopenNode, 2023-03-17:
It is fixed now, works also in Linux.

closed 2023-03-17 by CANopenNode