CANopenNode/CANopenEditor #46: Typo in generatd DCF file

thesummer, 2023-02-15:
Hello,

I used the CANopenEditor version 4.0-104-g6f50f73 to configure some nodes and generate a coresponding DCF.
In the DCF I noticed that the block with the general device setup is generated as `[DeviceComissioning]` while it should be `[DeviceCommissioning]` (with "mm").

It's a small type, but it prevents for example https://canopen.readthedocs.io/en/latest/ to read in the generated DCF.
I looked for the string in the source code and fix it myself, but couldn't find it.

Do you have any suggestions? It seems to be one of the few parts where it is not possible to rename the field through the GUI.

trojanobelix, 2023-02-15:
Unfortunately, the standard (DSP302) defines the section with "DeviceCommissioning". Clearly a typo, but unfortunately defined exactly like this. 

This is why the file is written in conformity with the standard. A tool that evaluates this should therefore at least accept this way of writing.

thesummer, 2023-02-17:
Oh, that is a weird thing. Thanks for the fast reply. I will try to fix it in the Python module then.

trojanobelix, 2023-02-17:
DSP306  Version 1.1 Date: 29.06.2001 stated
5.3 Device Co**mm**issioning
There is an additional section in the DCF named DeviceCo**m**issioning:
(...)
Example:
[DeviceCo**m**issioning]

thesummer, 2023-02-20:
I just checked the version 1.3.0 from 2005. The typo with *DeviceComissioning* is still there, but I think I found a smaller thing which is on the CANopenEditor side.

> There shall be an additional section in the DCF named _DeviceComissioning_:
> _NodeID_ shall indicate the device’s address (Unsigned8)
> _NodeName_ shall indicate the node name (max 246 characters)
> _Baud**r**ate_ shall indicate the device’s baudrate (Unsigned16)
> _NetNumber_ shall indicate the number of the network (Unsigned32)
> _NetworkName_ shall indicate the name of the network (max 243 characters)
> 

In a generated DCF with CANopenEditor I got 
```
[DeviceComissioning]
NodeId=1
NodeName=
BaudRate=500
NetNumber=0
NetworkName=
```
So, _NodeI**D**_ and _Baud**r**ate_ seem to be a little off with the capitalization.

thesummer, 2023-02-21:
Thank you for the fast fix, but where do I find the binaries? Is there a pipeline where I can download the artifacts?

trojanobelix, 2023-02-21:
> Thank you for the fast fix, but where do I find the binaries? Is there a pipeline where I can download the artifacts?

I release the binaries at irregular intervals, though I have been able to do at least a minimal test cycle.
You have received an email with the binaries

thesummer, 2023-02-21:
Hm, can't find any email with an archive or a download link in my mailbox. Only the notifications with your replies.

trojanobelix, 2023-02-21:
[Release_Bugfix_76baf61.zip](https://github.com/CANopenNode/CANopenEditor/files/10793910/Release_Bugfix_76baf61.zip)

thesummer, 2023-02-21:
Thanks you very much. I can confirm that this fixes the behavior. 

PS: I have got an email from microsoft that your email was put into quarantine. Seems like it flagged the binaries inside the archive.

closed 2023-02-21 by thesummer

trojanobelix, 2023-02-21:
/* Prevent WebKit and Windows mobile changing default text sizes */
    body,
    table,
    td,
    a {
      -webkit-text-size-adjust: 100%;
      -ms-text-size-adjust: 100%;
    }

    /* Remove spacing between tables in Outlook 2007 and up */
    table,
    td {
      mso-table-lspace: 0pt;
      mso-table-rspace: 0pt;
    }

    /* RESET STYLES */
    img {
      border: 0;
      height: auto;
      line-height: 100%;
      outline: none;
      text-decoration: none;
      -ms-interpolation-mode: bicubic;
    }

    table {
      border-collapse: collapse !important;
    }

    /* iOS BLUE LINKS */
    a[x-apple-data-detectors] {
      color: inherit !important;
      text-decoration: none !important;
      font-size: inherit !important;
      font-family: inherit !important;
      font-weight: inherit !important;
      line-height: inherit !important;
    }

    /* BASE STYLES */
    body {
      height: 100% !important;
      margin: 0 !important;
      padding: 0 !important;
      width: 100% !important;
      background: #E5E5E5;
    }

    h1,
    h2,
    h3,
    h4,
    table,
    td,
    th,
    a,
    p,
    span,
    div,
    i,
    em,
    li,
    ul,
    .img-block,
    img,
    .comment {
      font-family: 'Segoe UI', '-apple-system', 'BlinkMacSystemFont', 'Roboto', 'Arial', sans-serif;
    }

[... 586 lines cut ...]
            
          

          

          
            
              
                
                  

                    © 2023 Microsoft Corporation. All rights reserved.
                  
                
                
                  
                    
                      Privacy Statement
                  
                

                
                  
                    
                      Acceptable Use Policy

CANopenNode, 2023-02-24:
Hi,

In the latest standard (2019) there is still:

> 7.3.5 Device commissioning
> There shall be an additional section in DCF with the entry `DeviceComissioning`...

I think, this is intended, keyword in the standard can't just change.

thesummer, 2023-02-24:
I also just noticed that according to the standard the keys are not case sensitive. So strictly speaking this change wasn't even necessary. Sorry for the noise.

trojanobelix, 2023-02-25:
It's okay, it can be good that others don't fit the standard. It doesn't hurt and I still had some life time left. :-)