CANopenNode/CANopenEditor #47: Object Dictionary Editor: TPDO 0x1A00,0x01: not mappable OD entry 0x6000,0x00

ksdexory, 2023-03-06:
Following the CANOpen Node STM32 example I am getting the following error:
TPDO 0x1A00,0x01: not mappable OD entry 0x6000,0x00
https://www.youtube.com/watch?v=R-r5qIOTjOo
(creating a TPDO section at 1:26)

trojanobelix, 2023-03-07:
Which version do you use? Please use the latest release or the bugfix branch.

ksdexory, 2023-03-07:
I came across that in version v4.0-104-g6f50f73. I found a workaround, when I give transmit/receive access to PDO under the object dictionary tab -> object settings, the problem goes away. I was wondering if there is a bug on that field and the logic is the other way around (t gives access to receive and r gives access to t).
[image]

trojanobelix, 2023-03-07:
@CANopenNode : Janez, we had a discussion about the PDO access types and whether they are consistent across all import/export options. I have a halfway overview of the V1.x stuff with rww and so on, but with the new definitions (tr, etc) I lack the deeper insight.   Do you have a feeling for whether everything is OK? If not, I will try to dive deeper, but it may take some time.

CANopenNode, 2023-03-17:
I clarified myself about "rww" and "rwr". This is from the standard:
> The “rwr” indicates that the related object is read or is written using SDO, but is only mapped to TPDO.
> The “rww” indicates that the related object is read and is written using SDO, but is only mapped to a RPDO.

I think, it is now correct in CANopenEditor.

CANopenNode, 2023-03-17:
This should be fixed now.

closed 2023-03-20 by CANopenNode