CANopenNode/CANopenEditor #48: High/Low limit entries have no effect on the object dictionary?

sametoktn, 2023-03-15:
I reported the same problem to the CANopen Node project, a related project to this application. Since I haven't received an answer yet, I would also like to get information from here.

High/Low limit values ​​for objects have no effect on the output of the object dictionary. In other words, the min/max values ​​I entered do not cause the objects to be limited in any way. Is there a solution to this or will it be fixed if it is a bug?

trojanobelix, 2023-03-15:
You are correct, the EDS does not reflect the high/low limits. The high/low limits are optional according to CiA DS306. So it is not a bug. 
It seems to me that it could be useful information, but the standard does not specify a use. That's probably why it's only optional.

sametoktn, 2023-03-16:
Thank you for your interest, you are right when you say this subject that CiA has documented.

closed 2023-03-16 by sametoktn