CANopenNode/CANopenEditor #49: There is a bug with app UI

SparkShadow, 2023-03-28:
I download the [CANopenEditor-build.zip] source and run.But the UI in "Object Dictinoary" drives me carzy.
[image]
[image]
I don't know why it happend.

CANopenNode, 2023-03-28:
I also can't reproduce this. In my case works normally on different computers: Ubuntu 22.04, Windows 10, out of the box.
Maybe you can try another computer.

SparkShadow, 2023-04-06:
> 

[image]
About the xdd file,I load the file proviede by the app.And i have testd the file provided by the projected CanopenNode [https://github.com/CANopenNode/CanOpenSTM32](url),it seems to have the same problem.I have tried to change the resolution ratio,but it didn't work.

trojanobelix, 2023-04-06:
Please try it on a computer without a "windows feature experience pack"

trojanobelix, 2023-05-09:
No feedback, so close