CANopenNode/CANopenEditor #50: UI: TPDO and RPDO mapping does not show up the byte information

rohinidr, 2023-03-30:
[image]
[image]

trojanobelix, 2023-03-30:
Strange. Your version info is unknown. What version do you use? OS? XDD? Can you provide more detailed information?

rohinidr, 2023-03-30:
> Strange. Your version info is unknown. What version do you use? OS? XDD? Can you provide more detailed information?

I am using Windows 11 . Its virtual machine on MAC (Parallels to be more specific).

trojanobelix, 2023-03-31:
[image]

And the Editor version? Seems that you are not using the latest binaries. Should look like:

[image]

rohinidr, 2023-04-06:
After trying everything , I just figured out it was the issue with my display resolution ! I am able to see the byte information

closed 2023-04-06 by rohinidr