CANopenNode/CANopenEditor #51: Project saves always with the default data

killdaclick, 2023-04-14:
The issue is clearly visible in the following video:
https://youtu.be/V2HeW5CfOvE

The new project is created and multiple fields are changed in 'Device info' area.
After saving the project using File->Save Project As or File->Save Project all the data fields are reverted to factory values before save and the file is saved with reverted data.
It is not possible to save the Project anyhow. 

It works the same when running native on Windows, Linux on Mono and Linux on Wine.
Am I doing something wrong?

killdaclick, 2023-04-14:
Ok, so one needs to press the Update button before the save. It was not obvious :)

trojanobelix, 2023-04-14:
Self assigned: Check update behavior. Red color for update button on change of value.

trojanobelix, 2023-04-23:
Please try the latest bugfix release 93d61ccf2dd36146d190ef55339da57a1faad158

killdaclick, 2023-04-23:
Is it available as binary?

trojanobelix, 2023-04-24:
[Bugfix.zip](https://github.com/CANopenNode/CANopenEditor/files/11306819/Bugfix.zip)

trojanobelix, 2023-05-09:
No feedback, so close

closed 2023-05-09 by trojanobelix