CANopenNode/CANopenEditor #54: Errors when exporting to object dictionary

spydatron, 2023-04-28:
Hi, When i import an eds file for the maxon epos4 controller and export the CANopenNode OD.h & OD.c, i receive the following errors:

```
Error in 0x1000: 'Count Label' must be 'NMT'
Error in 0x1001: 'Count Label' must be 'EM'
Error in 0x1005: 'Count Label' must be 'SYNC'
Error in 0x1010: 'Count Label' must be 'STORAGE'
Error in 0x1011: SubIndexes in ARRAY must be in sequence!
Error in 0x1014: 'Count Label' must be 'EM_PROD'
Error in 0x1016: 'Count Label' must be 'HB_CONS'
Error in 0x1017: 'Count Label' must be 'HB_PROD'
Error in 0x1200: 'Count Label' must be 'SDO_SRV'
Error in 0x1400: 'Count Label' must be 'RPDO'
Error in 0x1401: 'Count Label' must be 'RPDO'
Error in 0x1402: 'Count Label' must be 'RPDO'
Error in 0x1403: 'Count Label' must be 'RPDO'
Error in 0x1800: 'Count Label' must be 'TPDO'
Error in 0x1801: 'Count Label' must be 'TPDO'
Error in 0x1802: 'Count Label' must be 'TPDO'
Error in 0x1803: 'Count Label' must be 'TPDO'
Error in 0x3000: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!
Error in 0x3000: Attributes of elements in ARRAY must be equal
Error in 0x3000: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!
Error in 0x3000: Attributes of elements in ARRAY must be equal
Error in 0x301A: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!
Error in 0x301A: Attributes of elements in ARRAY must be equal
Error in 0x3141: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!
Error in 0x3141: Attributes of elements in ARRAY must be equal
Error in 0x3150: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!
Error in 0x3150: Attributes of elements in ARRAY must be equal
Error in 0x31E4: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!
Error in 0x31E4: Attributes of elements in ARRAY must be equal
Error in 0x3248: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!
Error in 0x3248: Attributes of elements in ARRAY must be equal
Error in 0x60E5: SubIndexes in ARRAY must be in sequence!
Error in 0x60E5: SubIndexes in ARRAY must be in sequence!
Error in 0x60E5: SubIndexes in ARRAY must be in sequence!
```

What could be the problem and how can i fix it?

spydatron, 2023-04-28:
This is the eds file:

```

;*****************************************************************
; maxon motor ag
; all rights reserved
;*****************************************************************

[FileInfo]
FileName=ObjectDictionary.eds
FileVersion=2
FileRevision=0
Description=Electronic Data Sheet for EPOS4 Module 50/15
EDSVersion=4.0

CreationTime=10:54AM
CreationDate=04-09-2020
CreatedBy=mmc

ModificationTime=10:54AM
ModificationDate=04-09-2020
ModifiedBy=mmc

[Comments]
Lines=4
Line1=Software Version:    0x0161
Line2=Hardware Version:    0x6552
Line3=Application Number:  0x0000
Line4=Application Version: 0x0000

[DeviceInfo]
VendorName=maxon motor ag
VendorNumber=0x000000FB
ProductName=EPOS4 Module 50/15
ProductNumber=0x65520000
RevisionNumber=0x01610000

OrderCode=000000

BaudRate_1000=1
BaudRate_800=1
BaudRate_500=1
BaudRate_250=1
BaudRate_125=1
BaudRate_50=1
BaudRate_20=1
BaudRate_10=0

SimpleBootUpMaster=0
SimpleBootUpSlave=1
Granularity=8
DynamicChannelsSupported=0
GroupMessaging=0

NrOfRXPDO=4
NrOfTXPDO=4
LSS_Supported=1

[DummyUsage]
Dummy0001=0
Dummy0002=0
Dummy0003=0
Dummy0004=0
Dummy0005=0
Dummy0006=0
Dummy0007=0

;*****************************************************************
; Object Dictionary
[... 5350 lines cut ...]
ObjectType=0x7
DataType=0x4
AccessType=rww
DefaultValue=0
PDOMapping=1
ObjFlags=1

[6402]
ParameterName=Motor type
ObjectType=0x7
DataType=0x6
AccessType=rw
DefaultValue=10
PDOMapping=0
ObjFlags=0

[6502]
ParameterName=Supported drive modes
ObjectType=0x7
DataType=0x7
AccessType=ro
PDOMapping=1
ObjFlags=1

```

trojanobelix, 2023-04-30:
[maxxon.eds.txt](https://github.com/CANopenNode/CANopenEditor/files/11360813/maxxon.eds.txt)

For the countlables I have to dig a little deeper

@CANopenNode Can you give me a hint for the other wrnings?
Do we need these Warnings in the CanOpenExporter?
There is nothing in the CiA 306 definition for EDS files that justifies these warnings! The EDSchecker from Vector would also check this object as error-free!

` if (data.cType != dataElem.cType || data.length != dataElem.length)
                            Warnings.AddWarning($"Error in 0x{indexH}: Data type of elements in ARRAY must be equal!", Warnings.warning_class.WARNING_BUILD);
 if ((data.cValue == null && dataElem.cValue != null) || (data.cValue != null && dataElem.cValue == null))
                            Warnings.AddWarning($"Error in 0x{indexH}: Default value must be defined on all ARRAY elements or must be undefined on all ARRAY elements!", Warnings.warning_class.WARNING_BUILD);
if (attr != attrElem)
                            Warnings.AddWarning($"Error in 0x{indexH}: Attributes of elements in ARRAY must be equal", Warnings.warning_class.WARNING_BUILD);
`

Robotae, 2023-08-11:
I've had a similar issue with generating .[ch] files from an .eds file. I believe it's caused by commit 5b7f3f5 from 16th March 2023. This requires the user to provide correct count labels; fine if you're using a .xpd file but no good for .eds files as they don't contain count labels.
This can be demonstrated by opening DS301_profile.xpd from the EDSEditorGUI/Profiles directory, export the .[ch] files (File->Export CanOpenNode...), export as an .eds file (File->Export...), closing the .xpd file, opening the .eds file and exporting the .[ch] files. OD.h exported from the .xpd file contains all the OD_CNT_* defines in the "Counters of OD objects" section but OD.h exported from the .eds file contains none of these.
I'm using EDSSharp on the command line to generate .[ch] files from an .eds file. How about adding an option to EDSSharp (and EDSEditor) to automatically add count labels if they're missing, as it used to do, so these tools can be used with .eds files again?

CANopenNode, 2023-11-10:
If you import eds file, there is no information about CANopenNode specific parameters. Also have to follow rules described here: https://github.com/CANopenNode/CANopenNode/blob/master/doc/objectDictionary.md

Or use communication objects from DS301_profile.xpd

trojanobelix, 2024-04-22:
Closed: No feedback

closed 2024-04-22 by trojanobelix