CANopenNode/CANopenEditor #55: Two (different) objects with the same name can be created but will not be exported in source code

trojanobelix, 2023-05-12:
If there are two (different) objects with the same name only one will be created in  sCO_OD_RAM
Without any error/warning. 

Alternitives:
1. It should create an ERROR message on export if not permitted (Compare with standard)
2. It should not be created if not permitted
3. It could add an index on export (if it is permitted by the standard)

[image]

`/*2002 */
        #define OD_2002_stringTest                                  0x2002

        #define OD_2002_0_stringTest_maxSubIndex                    0
        #define OD_2002_1_stringTest_subObject1                     1
        #define OD_2002_2_stringTest_subObject1                     2
        #define OD_2002_3_stringTest_subObject1                     3
        #define OD_2002_4_stringTest_subObject1                     4
        #define OD_2002_5_stringTest_subObject1                     5
        #define OD_2002_6_stringTest_subObject1                     6
        #define OD_2002_7_stringTest_subObject1                     7

/*******************************************************************************
   STRUCTURES FOR VARIABLES IN DIFFERENT MEMORY LOCATIONS
*******************************************************************************/
#define  CO_OD_FIRST_LAST_WORD     0x55 //Any value from 0x01 to 0xFE. If changed, EEPROM will be reinitialized.

/***** Structure for RAM variables ********************************************/
struct sCO_OD_RAM{
               UNSIGNED32     FirstWord;

/*2000      */ VISIBLE_STRING  stringTest[2];
/*2001      */ UNSIGNED32      newObject[5];

               UNSIGNED32     LastWord;
};`

CANopenNode, 2023-11-10:
It seems it only happens in CANOPENNODE_LEGACY exporter. V4 exporter adds index to the variable name, so there is no problem. Maybe just add error message to LEGACY exporter.

closed 2026-10-20 by robincornelius

robincornelius, 2026-10-20:
Resolved by merged PR #237: colliding legacy storage names are disambiguated with index suffixes, and incompatible layouts no longer silently share a storage field. Issue233LegacyDuplicateScalarNamesHaveUniqueStorage and the shared-record/vendor generated-C regressions cover this area. The merged code passed the four .NET Framework 4.8.1/.NET 8 Debug/Release CI jobs. This issue was closed automatically by the merge.