CANopenNode/CANopenEditor #57: $NODEID should not be case sensitive

trojanobelix, 2023-05-12:
$NODEID should not be case sensitive 
Convert to uppercase on loading or do not create warning

EDS Checker is fine with "$NodeID"
DS306 does not prohibit mixed case explicite:
For entries of one of the integer types it is allowed to use a formula. This gives the chance to describe values, that depend on other  values. For example, the COB-ID of the default PDOs depend on the Node-ID of the device.
The syntax of the formula is given by the following EBNF description: 
IntEntryValue = $NODEID { Ó+Ó number }
For concrete devices, $NODEID will be replaced by the actual Node-ID of the device. The $NODEID must appear at the beginning of the expression. Otherwise the line is interpreted as without a formula. Actually it is only possible to use an offset to this NodeID in the formula. More complex expressions are not allowed.

Ashenoneq, 2023-05-13:
Hi! How to export the EDS file? I did not find any interface.

trojanobelix, 2023-05-17:
Should be fixed with the last commit. Will be reopened if not.

closed 2023-05-17 by trojanobelix