CANopenNode/CANopenEditor #64: System.ArgumentException with grid columns

tbitcs, 2023-07-04:
I started getting a strange runtime error after adding a new object. The error I get is this:

```
System.ArgumentException: Grid has only 67 columns, you tried to insert cell into position 2;67.You should probably call Redim on grid to increase it's column size
   at SourceGrid.Grid.DirectSetCell(Position position, ICell cell)
   at SourceGrid.Grid.InsertCell(Int32 row, Int32 col, ICell p_cell)
   at SourceGrid.Grid.set_Item(Position position, ICell value)
   at SourceGrid.Grid.set_Item(Int32 row, Int32 col, ICell value)
   at ODEditor.DevicePDOView2.UpdatePDOinfo(Boolean updatechoices) in Z:\CANopen\CANopenEditor\EDSEditorGUI\DevicePDOView2.cs:line 462
   at ODEditor.DeviceView.dispatch_updatePDOinfo() in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceView.cs:line 80
   at ODEditor.MyTabUserControl.doUpdatePDOs() in Z:\CANopen\CANopenEditor\EDSEditorGUI\MyTabUserControl.cs:line 44
   at ODEditor.DeviceODView.PopulateObjectLists(EDSsharp eds_target) in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceODView.cs:line 162
   at ODEditor.DeviceView.dispatch_updateOD() in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceView.cs:line 92
   at ODEditor.ODEditor_MainForm.openXDDfile(String path) in Z:\CANopen\CANopenEditor\EDSEditorGUI\Form1.cs:line 345
System.ArgumentException: Grid has only 67 columns, you tried to insert cell into position 2;67.You should probably call Redim on grid to increase it's column size
   at SourceGrid.Grid.DirectSetCell(Position position, ICell cell)
   at SourceGrid.Grid.InsertCell(Int32 row, Int32 col, ICell p_cell)
   at SourceGrid.Grid.set_Item(Position position, ICell value)
   at SourceGrid.Grid.set_Item(Int32 row, Int32 col, ICell value)
   at ODEditor.DevicePDOView2.UpdatePDOinfo(Boolean updatechoices) in Z:\CANopen\CANopenEditor\EDSEditorGUI\DevicePDOView2.cs:line 462
   at ODEditor.DeviceView.dispatch_updatePDOinfo() in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceView.cs:line 80
   at ODEditor.MyTabUserControl.doUpdatePDOs() in Z:\CANopen\CANopenEditor\EDSEditorGUI\MyTabUserControl.cs:line 44
   at ODEditor.DeviceODView.PopulateObjectLists(EDSsharp eds_target) in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceODView.cs:line 162
   at ODEditor.DeviceView.dispatch_updateOD() in Z:\CANopen\CANopenEditor\EDSEditorGUI\DeviceView.cs:line 92
   at ODEditor.ODEditor_MainForm.openXDDfile(String path) in Z:\CANopen\CANopenEditor\EDSEditorGUI\Form1.cs:line 345
```
The issue can be duplicated when opening this file: [ds402_ort-bldc.zip](https://github.com/CANopenNode/CANopenEditor/files/11951621/ds402_ort-bldc.zip)

trojanobelix, 2023-07-05:
In 0x1a00 an attempt is made to map more than 8 bytes into a PDO.
Fix: Limit the length of the objects to be mapped in a PDO to the maximum length of a CAN message (8 bytes).

Please try the latest bugfix branch  6cee7706ef8c02c4f45e7eb68d2626692d175554

tbitcs, 2023-07-05:
This fixes it. Thanks!

closed 2023-07-06 by trojanobelix