CANopenNode/CANopenEditor #65: Question: are the generated C files subject to (L)GPL?

lohralfen, 2023-07-06:
It seems to depend per software stack whether or not the generated files are also subject to the same (L)GPL license.
Since I can't find any example output files in this repository, could you confirm or deny whether the output files are licensed, and if yes, by which license?

lohralfen, 2023-07-06:
Thanks for your response. Can you direct me to where it specifically says the generated files are also licensed under GPL 3.0? I could not find that part.

I indeed saw the Apache License, version 2. But as long as the editor itself is licensed under GPL, I guess that "trumps" this remark on the stack project page.

trojanobelix, 2023-07-06:
This follows IMHO from the GPL itself.

 _The Corresponding Source need not include anything that users can regenerate automatically from other parts of the Corresponding Source._

The XDD ist your own work and is sufficient to generate the source code. The generated code is not covered by the GPL

lohralfen, 2023-07-06:
This renders me confused based on: https://www.gnu.org/licenses/gpl-faq.en.html#WhatCaseIsOutputGPL

trojanobelix, 2023-07-07:
GPL FAQ stated:
_**The output of a program is not, in general, covered by the copyright on the code of the program. So the license of the code of the program does not apply to the output**, whether you pipe it into a file, make a screenshot, screencast, or video._

and that fits perfect to:
_The XDD ist your own work and is sufficient to generate the source code. **The generated code is not covered by the GPL**_

Okay, glad I could help.

closed 2023-07-07 by trojanobelix