CANopenNode/CANopenEditor #67: LSS Master/Slave parameters does not export to OD.c/h

killdaclick, 2023-08-01:
After switching the "LSS Master" or "LSS Slave" under Device Info->General and Master Features and generating the OD.c/h does not change the CNT_LSS_MST / CNT_LSS_SLV value in OD.h. It is always equal to zero as in below example:
```
(config).CNT_LSS_SLV = 0;\
(config).CNT_LSS_MST = 0;\
```

trojanobelix, 2023-08-02:
I can confirm that "LSS Master" and "LSS Slave" under Device Info has no influence on the OD files.
I have no experience with LSS so unfortunately I cannot assess whether and how this should be used in the stack/ OD files. 

Maybe Janez can contribute something.

killdaclick, 2023-08-03:
I suppose it just needs to set only those values to corresponding '1'.
When I have set this manually LSS is working both in Master and Slave. Those CNT_LSS_* are checked in the code.

CANopenNode, 2023-11-10:
That is true for CANopenNode v1.3

For V4 you have to use `CO_CONFIG_LSS` macro from https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_config.h

closed 2023-11-10 by CANopenNode

killdaclick, 2023-11-10:
Do you understand that the GUI is not doing that? It is not changing anything. You have to manually edit the files, use macro or whatever. The GUI has no effect on the config file.

CANopenNode, 2023-11-10:
Yes, of course. In V4 it is intentional, in LEGACY, you have to edit manually.