CANopenNode/CANopenEditor #69: Impossible to import .eds file from Weidmueller UR20-FBC-CAN

MartinHolcik, 2023-09-14:
Hello, I am getting these warnings when attempting to import [WI133489V1_25.eds](https://mdcop.weidmueller.com/mediadelivery/asset/900_//68246?_ga=2.50495983.750335046.1694688939-1581987833.1694688939) file (Weidmueller's UR20-FBC-CAN EDS):
[image]
Can you give me a hint what I am doing wrong or where can be the issue?
Many thanks for any inputs
Regards
Martin

trojanobelix, 2023-09-15:
I have corrected the module parsing and it should be okay now.
The module tab view is not implemented, so you will not see the modules. That needs to be fixed in future.
Please try the latest bugfix commit ad5da4c43001ff84901e9dbcc622350e34fef04d

trojanobelix, 2023-09-15:
@CANopenNode Do you know what happes to the module view?

trojanobelix, 2023-09-15:
[WI133489V1_25.eds.txt](https://github.com/CANopenNode/CANopenEditor/files/12617787/WI133489V1_25.eds.txt)

CANopenNode, 2023-11-10:
> @CANopenNode Do you know what happens to the module view?

I thought it is unimplemented, so I just disabled it, here: https://github.com/CANopenNode/CANopenEditor/commit/43592ba22ea9438d3f9987d60a371aa47d56c1d8

trojanobelix, 2024-07-17:
Think you are right 

 CiA Draft Standard Proposal 306:
"If several modules are gathered to form a new Sub-Index, then the number is 0, followed by semicolon and the
number of bits that are created per module to build a new Sub-Index"

This feature is not implemented yet, and the parser will crash if something like **Count=0;4** occurs. 

Example from WI133489V1_25.eds:

```
[M1SubExt6000]
ParameterName=ReadInput8Bit
ObjectType=0x7
DataType=0x0005
AccessType=ro
PDOMapping=1
Count=0;4
```

trojanobelix, 2025-12-02:
Closed, no feedback.

closed 2025-12-02 by trojanobelix