CANopenNode/CANopenEditor #71: Layout of PDO tab oversize out of window.

Spider84, 2023-10-17:
In design view and runtime the width and height of PDO grid1 is oversize main window.

[image]

[image]

trojanobelix, 2023-10-17:
The current display is a compromise of the font size and standard window size. When maximising the window, it will fit in most cases.

Other suggestions or PR that optimises the display are welcome.

Spider84, 2023-10-17:
I use default windows fonts and sizes. Default build generates oversized layout and maximized window don't fit them.

Spider84, 2023-10-18:
[image]
[image]
[image]

As you can see, layout change size (width) with main window size (width) but in design width of tableLayoutPanel1 must be more smaller to be compiled to rigth size.

And distance between Zoom buttons is increasing....

closed 2023-11-06 by trojanobelix