CANopenNode/CANopenEditor #73: "not specified" type for Domain objects

GRoussos, 2023-10-19:
I have a .xdd file that includes a read-only object with domain data type, which I use with a custom read extension function. With previous EDSedtor versions this was exported correctly, without the object being included in the OD. With the latest build version ( v4.1-30-g5935269 ) the Domain object is included in the OD C struct with the type as "not specified". I was able to fix it manually by removing the bad object definition from the OD struct and changing the original data pointer for it to NULL. Is there some special way to make EDSeditor not include the domain object in the OD C struct?

trojanobelix, 2023-10-19:
Can you provide the xdd and the "good" and "bad" OD.c?

Thanks

gbracht, 2023-11-08:
Hello @trojanobelix , @GRoussos ,

I've been having the same problem with using Domain type in the OD. I can't attach a "good" OD yet because I haven't fixed it yet, but I can attach a bad one. 
The attached OD files are from a blank project with the basic DS301 profile with one device profile specific parameter "New Object" at address 0x6000. I gave it domain type and in the OD.h file the type is listed as "not specified" so it doesn't compile. Any help would be greatly appreciated.
 Also if there is a specific commit that I can check out where the Domain type export was working that would be helpful as well.
[OD_c.txt](https://github.com/CANopenNode/CANopenEditor/files/13298827/OD_c.txt)
[OD_h.txt](https://github.com/CANopenNode/CANopenEditor/files/13298828/OD_h.txt)

gbracht, 2023-11-08:
Update.

I was doing a little poking around to try to make my stream work without changing the OD manually. It seems like anything with length of 0 has the type "not specified". To test this I set the manufacturer device name (entry 0x1008) to be an empty string, which has the dataLength attribute as 0. In the OD.h file its type became "not specified", which is the same as what happens when trying to use the Domain type.

My current fix is to set the OD editor object that I want as a domain type to be a uint8, then after exporting I change the OD.c file entry to have a length of 0 so that the length can be set by the sdo client or the extension read/write functions.

trojanobelix, 2023-11-09:
Hmmm, let's talk about the  manufacturer device name (entry 0x1008) That is a VISUAL_STRING ( even it is the same with other strings and Domain, as far as I can see in the code).

 Creating a string with length of 0 (on stack) makes no sence. You can not set that the length by the sdo client or the extension read/write functions, because there is no memory allocated (sizeof(string) is 0).

I think it it enought to throw a exeption or a message window, if a string has not a default value. 

In the legacy exporter this message comes up:   _Object 0x1008/00 A string must have a default value to set the required datasize for canopen node, i have set this to [1] byte to prevent compile errors_

mlout, 2023-11-09:
Hi @trojanobelix,

We are running into the same issue with data type VISIBLE_STRING, in our case with registers 0x1009 Manufacturer hardware version and 0x100A Manufacturer software version.

Like @GRoussos we are using OD extensions when these register are read, then we read the hardware and software version from a specific memory region.

When using the CANopenEditor version v4.1-30-g5935269 I have to set a default value for these registers (to prevent having code that doesn't compile or having to manually edit the OD.c / OD.h files), this causes the dataLength attribute also to be set for these registers in the OD.c file.

What I noticed then was that CO_SDOserver.c check the length of the data read against the length defined in the OD (CO_SDOserver_process.c line 786), except if the length is defined as 0 in the OD.
That would mean that for the version number register, we would need to pad the response with additional bytes, only to make it work with the SDO server.
E.g. we expect our maximum version number string to be "v999.999.999", normally it will be something like "v1.0.0", but we will always need to add extra null terminator bytes in order to send it.

So we would actually prefer the behaviour of the previous editor version of not having a default value, so that the dataLength can be 0.
Or would it be better adjusting the behaviour of CO_SDOserver.c to allow it to send string with a shorter length than defined in the OD?

Best regards,
Marco

trojanobelix, 2023-11-09:
@CANopenNode Janez, do you have any idea how to handle this? I think all string sizes are constant (strings not created on heap) so max. string lenght must be used when defining a default string.

gbracht, 2023-11-09:
@trojanobelix I agree with you that reasons to have a 0 length string are few and far between, I simply attempted to do that with visual string hoping that I could get Domain type functionality (dataLength = 0) without modifying the exported OD. The purpose of the comment that I made was to show that selecting any option with data length of 0 (including Domain type) in the Editor software resulted in an exported OD that does not compile. Since the CANopenNode SDO server defaults to using the datasize from the OD even if an extension is used, this makes it impossible to have datastream-like SDO communication without modifying the exported OD.

Thank You for the advice and quick replies!

GRoussos, 2023-11-09:
@trojanobelix for strings (or other data types, eg. Domain) with 0 length it would make sense to me to skip including them in the OD all together, as the application will need to define extensions for them. As @gbracht notes, the SDO server may need some rework to work in this scenario.
Adding a minimum length (indirectly done by setting a default value) is a workaround that can lead to some memory waste, as when using extensions the OD object may not be used at all.

CANopenNode, 2023-11-10:
Hello,
CANopenNode V4 allows undefined data size, which is quite useful with strings and necessary with domains. Actually data for any type of variable does not even need to be stored by object dictionary, application can take full control. See https://github.com/CANopenNode/CANopenNode/blob/master/doc/objectDictionary.md

The problem mentioned above appears with the latest build (v4.1-30). But previous build (v4.1-2) produces correct object dictionary. For quick check you can load file https://github.com/CANopenNode/CANopenDemo/blob/master/demo/demoDevice.xdd and export V4 files OD.h and OD.c. File contains exotic variables and exported result should be the same.

Problem with CANopenEditor is, it does not have clear internal database, so some "small" changes may be quite problematic. When I started adopting it for V4, I tried to keep editor as unchanged as possible, but made V4 exporter with great care.

Besides, standards are not clear in all details, so some parts are required by CANopenNode V4, for example warnings in https://github.com/CANopenNode/CANopenEditor/commit/c9419d124634e3353a5d6c73d03498093f0a1e97.

Probably the source of problem is https://github.com/CANopenNode/CANopenEditor/commit/7dae47c638372c3331da4516a492c13ed20bc7dd.

If we would like to make more perfect CANopenEditor, I would suggest to start from scratch. But this is another topic.

---
Janez

trojanobelix, 2023-11-10:
Hi Janez, think your right that https://github.com/CANopenNode/CANopenEditor/commit/7dae47c638372c3331da4516a492c13ed20bc7dd.  messed it up

CANopenNode, 2023-11-10:
This is probably from issue #54, which wasn't really an issue. Sorry for my late answer there.

trojanobelix, 2023-11-11:
@CANopenNode Janez, do we agree that the https://github.com/CANopenNode/CANopenEditor/commit/7dae47c638372c3331da4516a492c13ed20bc7dd should be reverted?

trojanobelix, 2023-11-12:
done in bugfix branch (untested)

closed 2024-03-16 by trojanobelix