CANopenNode/CANopenEditor #74: Hello, about using CANopenEditor configuration SRDO problem

WeiShengxiujing, 2023-10-25:
Hello recently I have been using CANopenNode and CANopenEditor.I want to use the CiA304 function of CANopenNode，but I found that CANopenEditor It seems that it cannot be directly configured SRDO。In the 304 protocol，configure the SRDO communication parameter cob-id to be 0x1301-0x1340，the mapping parameters are 0x1381-0x13C0， 0x13 FEConfigure valid flags，0x13FF CRC。
When I use CANopenEditor insert profile，in DS301_profile.xpd dont have 0x13xx.
Now my problem is: Should I create the relevant SRDO configuration myself if I want to use SRDO? For example, creating a 0x1301 but I don't know the relevant sub indexes in 0x1301。May I ask how you can solve this problem.Please give me some guidance .Thank you very much

trojanobelix, 2023-10-30:
I do not have access to CiA-304 or EN 50325-5, but the necessary objects can be created in the CANopenEditor. 
When all relevant CiA-304 objects have been created, this file can be saved as a profile ("Save as: DS304_profiel.xpd). Please feel free to post it here or send it as PR.

WeiShengxiujing, 2023-11-02:
I am very sorry for only replying to you today and thank you very much for your guidance. I will establish the relevant configuration for 304 in CANopenEditor and export it as OD. c before attempting to use it. However, this work may take some time because I learned CANopenNode on STM32F407 first. What I am currently working on is porting CANopenNode to TI's TMS320F2375D. Although the code can run, there are still many issues. If I can succeed, I will share it. Thank you very much for replying to me during my busy work.

CANopenNode, 2023-11-10:
CiA-304 is not yet updated for use with CANopenNode V4. Files are there, but are disabled.

closed 2024-08-14 by CANopenNode