CANopenNode/CANopenEditor #75: An error occurred while I was editing PDO obiect dictionary, such that I lose my modification.

QingnanZhong, 2023-12-01:
System.NullReferenceException: 未将对象引用设置到对象的实例。
   在 libEDSsharp.PDOHelper.build_PDOlist(UInt16 startcob, List`1 slots) 位置 C:\Git_lokal\CANopenEditor\libEDSsharp\PDOHelper.cs:行号 219
   在 ODEditor.DevicePDOView2.UpdatePDOinfo(Boolean updatechoices) 位置 C:\Git_lokal\CANopenEditor\EDSEditorGUI\DevicePDOView2.cs:行号 432
   在 ODEditor.DeviceView.dispatch_updatePDOinfo() 位置 C:\Git_lokal\CANopenEditor\EDSEditorGUI\DeviceView.cs:行号 97
   在 ODEditor.MyTabUserControl.doUpdatePDOs() 位置 C:\Git_lokal\CANopenEditor\EDSEditorGUI\MyTabUserControl.cs:行号 44
   在 ODEditor.DeviceODView.PopulateObjectLists(EDSsharp eds_target) 位置 C:\Git_lokal\CANopenEditor\EDSEditorGUI\DeviceODView.cs:行号 189
   在 ODEditor.DeviceView.dispatch_updateOD() 位置 C:\Git_lokal\CANopenEditor\EDSEditorGUI\DeviceView.cs:行号 109
   在 ODEditor.ODEditor_MainForm.openXDDfile(String path) 位置 C:\Git_lokal\CANopenEditor\EDSEditorGUI\Form1.cs:行号 368

trojanobelix, 2023-12-07:
Can you provide something to reproduce? XDD, EDS, how to?

trojanobelix, 2024-03-16:
No feedback. Closed

closed 2024-03-16 by trojanobelix