CANopenNode/CANopenEditor #76: Unsigned32 Variable in TPDO doesn't have proper data length in bits

oliver99supergm, 2023-12-11:
Hello,

I created a TPDO containing a UNSIGNED32 variable  but the TPDO mapping parameter shows 16 bits not 32 as it should.  I've attached screen shots of the variable definition, TPDO graphical representation, and 0x1A01 object showing that the 3rd application object ends in 0x10 not 0x20.  I've also attached a zip containing the XDD file and the OD.c and OD.h that it generated.

[image]
[image]
[image]

[2700-002.zip](https://github.com/CANopenNode/CANopenEditor/files/13639840/2700-002.zip)

trojanobelix, 2023-12-12:
Sorry, I can't reproduce it.
Could it be that the value in 0x1A01 has been changed manually?

When I use the "TX_Mapping" tab and select a row (to activate the "Save" button), the value is correct after saving. 

Please try the latest bugfix branch.

oliver99supergm, 2023-12-12:
> CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thanks for the reply.

I didn't modify the 0x1A01 entry manually.  I changed the size of the 0x34C1 Battery Voltage from UNSIGNED16 to UNSIGNED32 and saved the results.  I then went to the "TX_Mapping" tab and confirmed that the variable change was reflected in the message showing the battery voltage as a four byte variable.  But the x1A01 entry was still showing 16 bits.

I just tried the following:

  1.  Go "Object Dictionary" tab.
  2.  Change 0x34C1 Battery Voltage from UNSIGNED16 to UNSIGNED32.
  3.  Confirm that 0x1A01 is still showing 16 bits for Application Object 3.
  4.  Go to "TX PDO Mapping" tab
  5.  Delete Battery Voltage variable from TPDO with index 0x1801.  Prior to deletion the message was showing the correct 4 byte variable size.
  6.  Add the Battery Voltage variable back to the message.  Again it is showing as proper 4 byte variable size.
  7.  Go to "Object Dictionary" tab.
  8.  Object 0x1A01 is now showing proper variable size as shown below:
[A screenshot of a computer  Description automatically generated]

Is it possible the variable size change doesn't propagate thru the dictionary properly?

trojanobelix, 2023-12-13:
Perhaps it was simply forgotten to save this change via the save button after adding the new 32-bit "Battery voltage" in the TX PDO mapping? Then when switching to tab Object directory the 16 bit value is still active in 0x1A01/3.

If I do not save in the TX PDO mapping and then save the entire project, I have exactly the described behaviour after reopening. 

I will try to add a warning or something similar when ready assigned PDO data in the object dictionary changes. 
But that doesn't sound trivial.

oliver99supergm, 2023-12-13:
Hello,

What caused the problem was changing the size of a variable already assigned to a TPDO.  The variable was 16 bits and already in use in the TPDO.  My only change was to change the variable size to 32 bits in its OD entry.  When I checked the the TX PDO Mapping page it showed the proper variable size in the graphics at the bottom of the screen but the corresponding mapping OD entry still showed the original 16 variable field size.  I didn't change anything in the TX PDO Mapping page.  I did make a variable size change that affected the TX PDO Mapping page and should have affected the corresponding TPDO mapping OD entry.  Are you saying I have to save the variable size change then go to the TX PDO Mapping page and hit save again there?

If I delete this variable from the TPDO, change its size, then add it back again to the same TPDO the corresponding mapping OD entry is now correct.

I think somehow the variable size being changed isn't being propagated completely when its already assigned to a TPDO prior to changing its size.

trojanobelix, 2023-12-14:
> Are you saying I have to save the variable size change then go to the TX PDO Mapping page and hit save again there?

Yes: save the variable size change then go to the TX PDO Mapping page **and just select a row** (to activate the "Save" button), the value in 0x1A01/3 is correct after saving.

The mapped objects are only analysed again with "Save" and the mapping is generated.

oliver99supergm, 2023-12-14:
Hello,

That was it.  I took a variable from 8 to 16 then followed your procedure below and the mapping OD entry was properly updated.  I then changed it back from 16 to 8 again following procedure below and it returned to its original form.

Thanks for the help on this - Bruce

closed 2024-10-09 by CANopenNode