CANopenNode/CANopenEditor #78: EDS file seems to generate wrong access type for PLC TPDO and RPDO

Jan-Jelle, 2024-01-05:
I am running into a problem where the EDS file has the wrong acces type for indexes in the TPDO and RPDO.

example: I set a variable called T1S to rw access, and it is used for the RPDO.

[image]
The EDS file now generates this result: access type rwr
[image]
the opposite happens for TPDO access:
[image]
result in EDS: rww
[image]

Codesys can only map the rwr access type to a TPDO and rww to a RPDO, so inverted with what i need. below the settings for an RPDO, where only the index variables for the TPDO's are accessible.
[image]

and here the opposite for the RPDO's:
[image]

to it seems that the generated EDS file creates the wrong access type with the variables. any idea why and how to solve?

trojanobelix, 2024-01-16:
Which version of the CANnodeEditor do you use?
Can you provide the XDD and the generated EDS files, please.

Jan-Jelle, 2024-01-23:
Hi,

I use the version from HamedJafar: https://github.com/HamedJafarzadeh/CANopenEditor/tree/742c4f3cceab78b3a2f79053491b983184505fab
Version 4.0.2-Alpha 
[image]

See EDS below.
[DS301_profile issue rwr TPDO.zip](https://github.com/CANopenNode/CANopenEditor/files/14028783/DS301_profile.issue.rwr.TPDO.zip)

Thanks for any help!

trojanobelix, 2024-01-24:
sorry, I am not able to investigate possible bugs in other forks. Please contact Hamed directly or try to reproduce this bug on this code base.

Thank you