CANopenNode/CANopenEditor #79: PDO Mapping 'Invalid' not work while 'Export CanOpenNode'

JarvisXing, 2024-02-19:
If check TX/RX PDO Mapping Invalid checkbox, COB in TX/RX PDO Mapping will change.
But COB-ID in Object Dictionary will not change. COB_IDUsedByTPDO is same as Object Dictionary in OD.c，
so PDO->valid is always zero.
Is it a bug or design?

trojanobelix, 2024-02-19:
Seems to work as expected (Bugfix branch):

[image]
[image]

`    },
    .x1800_TPDOCommunicationParameter = {
        .highestSub_indexSupported = 0x06,
        .COB_IDUsedByTPDO = 0x80000180,
        .transmissionType = 0xFE,
        .inhibitTime = 0x1388,
        .compatibilityEntry = 0x00,
        .eventTimer = 0x03E8,
        .SYNCStartValue = 0x00
    },`

JarvisXing, 2024-02-23:
thank you, my mistake. I didn't choose item.

closed 2024-02-26 by trojanobelix