CANopenNode/CANopenEditor #88: TPDO and RPDO Assigned COB IDs

RnR336, 2024-03-19:
Using version 4.1-30 of the editor, when I add a TPDO, the COB ID starts with 0xC000xxxx and a RPDO COB ID starts with 0x8000xxxx, even when I select the legacy exporter. Why is that when the actual COB is 11 bits? Thanks for the insight.

trojanobelix, 2024-03-20:
Think you mean 0x14xx/01 RPDO and 0x18xx/01 TPDO communication parameter:

**COB-ID used by RPDO:**
  * **bit 31: If set, PDO does not exist / is not valid**
  * bit 11-30: set to 0
  * bit 0-10: 11-bit CAN-ID
* Transmission type:
  * Value 0-240: synchronous, processed after next reception of SYNC object
  * Value 241-253: not used
  * Value 254: event-driven (manufacturer-specific)
  * Value 255: event-driven (device profile and application profile specific)
* Event timer in ms (0 = disabled) for deadline monitoring.

**COB-ID used by TPDO:**
  * **bit 31: If set, PDO does not exist / is not valid**
  * **bit 30: If set, NO RTR is allowed on this PDO**
  * bit 11-29: set to 0
  * bit 0-10: 11-bit CAN-ID
* Transmission type:
  * Value 0: synchronous (acyclic)
  * Value 1-240: synchronous (cyclic every (1-240)-th sync)
  * Value 241-253: not used
  * Value 254: event-driven (manufacturer-specific)
  * Value 255: event-driven (device profile and application profile specific)
* Inhibit time in multiple of 100µs, if the transmission type is set to 254 or 255 (0 = disabled).
* Event timer interval in ms, if the transmission type is set to 254 or 255 (0 = disabled).
* SYNC start value
  * Value 0: Counter of the SYNC message shall not be processed.
  * Value 1-240: The SYNC message with the counter value equal to this value shall be regarded as the first received SYNC message.

RnR336, 2024-03-22:
Yes, those are the OD values which the editor writes to when a new RPDO or TPDO is created from their respective tabs. Why is the 1 the default setting for bits 31 and 30 TPDOs and 1 for bit 31 for RPDOs?

trojanobelix, 2024-03-24:
why not? Makes sense to me, because you do not know the CAN-ID when creating the object. Not valid seems to be the right state at time of creating.

RnR336, 2024-03-25:
I see your point. When the user selects "Add new PDO" from either pane, the entry is added to the dictionary right then, but it is up to the user to fill in the remaining information for the PDO in a separate step. Even though I typically do this, it doesn't mean that's what users do and to ensure proper values I can see why that is the default behavior. Thanks for helping me understand the behavior.

closed 2024-03-25 by RnR336