CANopenNode/CANopenEditor #102: refactor CanopenEditor

nimrof, 2024-04-02:
Hi,
Feel free to change/add/subdivide/remove or "take" tasks, i just needed to have a place to talk/plan outside of the pull requests.

So this is things we agree we want to do? and this is high priority:
- [ ] Remove CanOpenNodeXML (inc bridge) & XDD v1
- [ ] Move import & export part of eds away from eds.cs
- [ ] Major undecided change/rewrite to the rest of eds.cs?

Suggestions in no order. low priority and it would be nice to talk about some of the big things:
- [ ] A uniform way do interact with importer/exporter
- [ ] More tests, close to 100% coverage on libEDSSharp
- [ ] Clean build
  build without warnings & messages from compiler
  (i want to continue on this to learn how it all works + its very easy to do if i have some spare time)
- [ ] Divide the DocumentationGen into markup & html
- [ ] NetworkPDOreport has some of the same code as DocumentationGen
  that would be nice to remove the need for the same code multiple places
- [ ] Native .net core GUI
  So we can move away from .net 4.8.1 while continue to support non-windows with a gui

CANopenNode, 2024-04-02:
I agree with all the tasks, that would be a great improvement and a base for potential future growth of the project.

trojanobelix, 2024-04-03:
Me too, on all tasks. I'm trying to reduce the number of warnings and messages. Most seem less relevant to me, but some are, e.g. "expression always true" in an if statement. 

Good idea to continue the discussion here.

CANopenNode, 2024-06-10:
[ ] Enable "Online" mode, so CANopenEditor could read or write Object Dictionary values on actual devices.

CANopenLinux in Raspberry PI, for example, has gateway capabilities with ascii communication via (TCP) socket. CANopenEditor should be able to use tcp socket and update actual values on all OD variables. Here is python example for local socket, but TCP socket works too: https://github.com/CANopenNode/CANopenLinux?tab=readme-ov-file#accessing-ascii-command-interface-with-python