CANopenNode/CANopenEditor #106: Can't add item into OD

jow905, 2024-04-08:
If the OD is empty, I can't add a item to it by right click my mouse,  but if it has one or more items, I can add more by right click one item that already in the list. I think this is the issue of the GUI.
The version of GUI I used is : V4.2-37-g9dc739c

[image]
[image]

nimrof, 2024-04-09:
Hi @jow905 
That is confusing i know.

You have to click the header with the normal/left mouse button to get the correct pop-up menu
[image]

jow905, 2024-04-12:
Hi， @nimrof Could you tell me what version you used? I try to click the header, but I still can't get the pop-up menu. If you use the same version with me, I think maybe the operate system is the problem, my OS is windows11 23H2, donet framework4.8.1.

nimrof, 2024-04-12:
Sorry, i was not on the latest version.
Running the same os&patch&framwork as you i can reproduce the problem.
I will debug

nimrof, 2024-04-12:
@jow905 
It was removed some time ago, not sure how it should be done without it.
Added it back and you can test the quickfix in the link below, will report back when i got something.
https://github.com/CANopenNode/CANopenEditor/actions/runs/8662518321/artifacts/1408698799

closed 2024-04-13 by nimrof

nimrof, 2024-04-13:
@jow905 
new release with the fixes can be found here:
https://github.com/CANopenNode/CANopenEditor/releases/tag/v4.2.1

jow905, 2024-04-13:
@nimrof , I test the new version, it works, thanks!

> @jow905 new release with the fixes can be found here: https://github.com/CANopenNode/CANopenEditor/releases/tag/v4.2.1