CANopenNode/CANopenEditor #110: CanopenEditor: Exporting to XDD v1.0 format sets wrong type on "Highest sub-index supported"

hepr-skylotec, 2024-04-24:
When using ARRAY types and exporting to XDD v1.0 "old" format, the 0x00 index of the arrays "Highest sub-index supported" gets the same type as the array elements in index 1 and up.

For SDOs of ARRAY type, the 0x00 index "Highest sub-index supported" is always an UNSIGNED8, regardless of the type used for the array elements. The editor handles this correctly in code generation and XDD v1.1 format but wrong in XDD v1.0 format export.

The `CANopenObject` for subindex 0 has the `dataType` attribute set to the same value as the rest of the subindex entries in the array. The `dataType` attribute should always be `0005` whic is for UNSIGNED8

CanopenEditor version : v4.2.0-0-g3735e28 on Windows 11

Current output:
```
          <CANopenObject index="220B" name="Temperatures" objectType="8" lowLimit="" highLimit="" actualValue="" denotation="" edseditor_extenstion_storagelocation="RAM" edseditor_extension_notifyonchange="false" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b" subNumber="9">
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0000" name="Highest sub-index supported" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="no" uniqueIDRef="UID_PARAM_220b00" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0001" name="FET Temperature Discharge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b01" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0002" name="FET Temperature Charge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b02" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0003" name="Cell Temperature - AFE 1" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b03" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0004" name="Cell Temperature - AFE 2" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b04" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0005" name="Cell Temperature - Gauge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b05" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0006" name="Internal Temperature AFE" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b06" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0007" name="Internal Temperature Fuel Gauge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b07" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0008" name="Internal Temperarure MCU" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b08" />
          </CANopenObject>
```

Expected output:
```
          <CANopenObject index="220B" name="Temperatures" objectType="8" lowLimit="" highLimit="" actualValue="" denotation="" edseditor_extenstion_storagelocation="RAM" edseditor_extension_notifyonchange="false" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b" subNumber="9">
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0000" name="Highest sub-index supported" objectType="7" dataType="0005" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="no" uniqueIDRef="UID_PARAM_220b00" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0001" name="FET Temperature Discharge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b01" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0002" name="FET Temperature Charge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b02" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0003" name="Cell Temperature - AFE 1" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b03" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0004" name="Cell Temperature - AFE 2" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b04" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0005" name="Cell Temperature - Gauge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b05" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0006" name="Internal Temperature AFE" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b06" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0007" name="Internal Temperature Fuel Gauge" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b07" />
            <CANopenSubObject edseditor_extension_notifyonchange="false" subIndex="0008" name="Internal Temperarure MCU" objectType="7" dataType="0008" lowLimit="" highLimit="" accessType="ro" actualValue="" denotation="" PDOmapping="TPDO" uniqueIDRef="UID_PARAM_220b08" />
          </CANopenObject>
```

nimrof, 2024-04-27:
Thanks for reporting the bug @hepr-skylotec,
It var very easy to reproduce with the information you gave us.

Got a bugfix ready if you want to test it out now:
https://github.com/CANopenNode/CANopenEditor/actions/runs/8859150670/artifacts/1453226221

trojanobelix, 2024-11-11:
What is the current status? Close it?

hepr-skylotec, 2024-11-12:
It looks like this is still broken in v4.2.2-0-g07966bd

trojanobelix, 2024-11-14:
@hepr-skylotec Please test if it solves the problem.

PR #111 (2024-04-27): 110 canopeneditor exporting to xdd v10 format sets wrong type on highest sub index supported
Bugfix + test

information about the bug can be found in the issue #110 

Resolved #110
closed 2024-11-18 by trojanobelix