CANopenNode/CANopenEditor #128: Missing definitions for subindex in CANOpenNode export

C0ntroller, 2024-07-26:
Hello,

When comparing old code that was generated using v0.6 of  [libedssharp](https://github.com/robincornelius/libedssharp) and my now newly generated one using this fork, I noticed, that there are definitions missing for accessing fields at a subindex. 

What I mean with this, is that for example I have an array at 0x2121 (lets call it Test_Array). Every one of the 14 fields inside (excluding the max sub-index entry) are properly named and defined (lets say, they are named Test_x where x is the index in hex).
But when exporting it, to directly access them, I would need to hardcode their sub-index.
For example:

```c
// For the 13th element
OD_ENTRY_H2121_Test_Array[0xd] = 1;
// Expected, like it was in previous versions
OD_ENTRY_H2121_Test_Array[ODA_Test_Array_Test_d] = 1;
```

Thsi change seems to already had happened at v0.8 of the original project.

Would it be possible to get these definitions back? 
It does not have to be exactly the same way, but any way would be great.

If not, what is the expected way to do this? Just hardcoding it?

nimrof, 2024-07-26:
Hi @C0ntroller,
There is something similar in the current version, but it is not hex as far as i can see

```
#define OD_1003_0_preDefinedErrorField_maxSubIndex          0
#define OD_1003_1_preDefinedErrorField_standardErrorField   1
#define OD_1003_2_preDefinedErrorField_standardErrorField   2
#define OD_1003_3_preDefinedErrorField_standardErrorField   3
#define OD_1003_4_preDefinedErrorField_standardErrorField   4
#define OD_1003_5_preDefinedErrorField_standardErrorField   5
#define OD_1003_6_preDefinedErrorField_standardErrorField   6
#define OD_1003_7_preDefinedErrorField_standardErrorField   7
#define OD_1003_8_preDefinedErrorField_standardErrorField   8
#define OD_1003_9_preDefinedErrorField_standardErrorField   9
#define OD_1003_10_preDefinedErrorField_standardErrorField  10
```

If elements are named this is generated also and that is kind of what you use, but it is name instead of raw index

```
/*1011, Data Type: UNSIGNED32, Array[4] */
#define OD_restoreDefaultParameters_idx                     0x1011
#define OD_restoreDefaultParameters                         CO_OD_RAM.restoreDefaultParameters
#define ODL_restoreDefaultParameters_arrayLength            4
#define ODA_restoreDefaultParameters_restoreAllDefaultParameters 0
#define ODA_restoreDefaultParameters_restoreCommunicationDefaultParameters 1
#define ODA_restoreDefaultParameters_restoreApplicationDefaultParameters 2
#define ODA_restoreDefaultParameters_restoreManufacturerDefinedDefaultParameters 3
```

> Would it be possible to get these definitions back? It does not have to be exactly the same way, but any way would be great.
I think so just make a pr. 

What do you think @CANopenNode ?

> If not, what is the expected way to do this? Just hardcoding it?

I mostly use the named index and hardcode or use the named version for subindex, but Janez could maybe tell the intended way

CANopenNode, 2024-07-26:
This is v1.3 exporter. As I remember, that was not removed, but names of subelements in the array must be unique, otherwise "ODA_" macros will not be generated.

C0ntroller, 2024-07-26:
> Hi @C0ntroller, There is something similar in the current version, but it is not hex as far as i can see
>
> ```
> #define OD_1003_0_preDefinedErrorField_maxSubIndex          0
> #define [...]

I can't really duplicate that.

Just so we mean the same thing, here's my workflow:
- Using the latest release of this repo (v4.2.1)
- Create a new product/new OD
- Insert profile -> DS301_profile.xpd
- Check everything and insert
- File -> Export CANOpenNode

After that I cant find anything like `#define OD_1003_1_preDefinedErrorField_standardErrorField   1` in the generated header (nor the .c but thats expected).

Am I at fault here?

> This is v1.3 exporter. As I remember, that was not removed, but names of subelements in the array must be unique, otherwise "ODA_" macros will not be generated.

In workflow above this this should be the case for the `Store parameters` field at 0x1010. But I still can't find the definitions.

nimrof, 2024-07-26:
> Am I at fault here?

maybe, is v4 exporter selected in preference?
If yes, select the other one and try again
[image]

CANopenNode, 2024-07-26:
V4 does not provide those macros. It works differently.

C0ntroller, 2024-07-30:
> maybe, is v4 exporter selected in preference? If yes, select the other one and try again [...]

Yes, this works, thank you.  
But is there any reason this was removed in the new version?

I know this is off topic, but I guess you are way more experienced than me using CANOpenNode: Should I rather use an "old" version rather than the current state (master says it's version 4.0)?

CANopenNode, 2024-07-30:
Make your own definitions? See also CANopenDemo.

nimrof, 2024-08-31:
Closing as non-active

closed 2024-08-31 by nimrof