CANopenNode/CANopenEditor #132: Support ESI for Ethercat

mgiaco, 2024-08-13:
Hi, I a wondering if there is any plan to support also ESI Format for Ethercat in the future?

nimrof, 2024-08-13:
Hi @mgiaco,
No official plan to support ESI format or anything related to Ethercat.
This is the closest you get to a official plan: https://github.com/CANopenNode/CANopenEditor/issues/102

The internal/common format used in the project uses CanOpen EDS so AFIK it would be a big job to support Ethercat

CANopenNode, 2024-08-14:
It depends, which information ESI file needs. According to [wikipedia](https://en.wikipedia.org/wiki/EtherCAT), Ethercat may use CANopen device profiles, so it may be possible to generate some sort of ESI file from existing data.

I don't have access to Ethercat [specifications](https://www.ethercat.org/en/downloads/downloads_6A46D45EA33C47ECB2BB2686BBA963EC.htm), but someone with access could:

- verify if current information from CANopenEditor are enough for ESI file, see for example recent [protobuffer](https://github.com/CANopenNode/CANopenEditor/blob/main/libEDSsharp/proto/CanOpen.proto) definition.
- Get xml schema for ESI file (seems, it is available). From xml schema cs file can be automatically generated, similar as https://github.com/CANopenNode/CANopenEditor/blob/main/libEDSsharp/CanOpenXSD_1_1.cs
- Create importer and exporter, similar as (not the newews version) https://github.com/CANopenNode/CANopenEditor/blob/1d123ecf13ff69d4aae3e5ee22be745bcbad3599/libEDSsharp/CanOpenXDD_1_1.cs

closed 2024-10-09 by CANopenNode

mgiaco, 2024-12-02:
Sorry i forgot to write back. Many thanks for your help. I will check in the future.