CANopenNode/CANopenEditor #138: Unused branches

CANopenNode, 2024-10-09:
I would like to clean some unused or obsolete branches in this repo. There are three such branches:

1. "Net4.8" - I made this branch before upgrade to net6 started. There is already a tag "v4.2". We have v4.8 builds in the main branch anyway. Safe to remove it.
2. "upgrade-to-net6" - @trojanobelix, can you please check, if there is something useful, otherwise I suggest to remove it. 
3. "build" - that was my "invention". There are some old binaries, from the time, we didn't have automated builds. I suggest to remove that part from the repo to avoid confusion.

trojanobelix, 2024-10-09:
Nothing useful from my side ;-) upgrade-to-net6 can be removed

trojanobelix, 2024-10-09:
"build" can be confusing and should be removed (IMHO)

CANopenNode, 2024-10-10:
Three obsolete branches are removed now and also readme is updated. I think, there is no more reference to build branch anywhere.
Hint: To cleanup local repository use: `git remote prune origin` and `git gc`

closed 2024-10-10 by CANopenNode