CANopenNode/CANopenEditor #139: Wine running on MacOS

ruziev-dev, 2024-10-23:
I'm using MacOS for development.
CANopenEditor GUI application works good through the Wine.

But CLI tools are not available.

```bash
# test command
wine cmd /c CANopenEditor/EDSEditor.exe --infile something.xdd --outfile something.eds
```

The command above load and run GUI program

```
0230:fixme:mscoree:parse_supported_runtime sku=L".NETFramework,Version=v4.8.1" not implemented
0230:fixme:mscoree:parse_supported_runtime sku=L".NETFramework,Version=v4.8.1" not implemented
0230:fixme:ntdll:NtQuerySystemInformation info_class SYSTEM_PERFORMANCE_INFORMATION
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Apple Braille.ttf"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\Baghdad.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\Gujarati Sangam MN.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\Gurmukhi Sangam MN.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\HelveticaNeue.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\Hoefler Text Ornaments.ttf"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\ITFDevanagari.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\ITFDevanagari.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\Kailasa.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\KohinoorGujarati.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\KufiStandardGK.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\Nadeem.ttc"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\Supplemental\\Skia.ttf"
0230:fixme:font:find_matching_face Untranslated charset 255
0230:fixme:font:get_nearest_charset returning DEFAULT_CHARSET face->fs.fsCsb[0] = 00000000 file = L"\\??\\Z:\\System\\Library\\Fonts\\ZapfDingbats.ttf"
```

I need it to automate project builds.
Does anybody know how to automatically export CANopenNode files from different `xdd` or `xpd` files with Wine?

nimrof, 2024-10-23:
Hi,
Sorry, but i do not have a mac to test it on, but the CLI should work without wine if it is compiled with .net 6.

Is that something you could try?
You will need to build the cli tool with .net6 as the cli tool is not included with the binary release files (that i just realized was a bad idea)

ruziev-dev, 2024-10-24:
@nimrof 
I'm not good on .net.
Could you help me with correct command to run?

I have downloaded [v4.2.2 from Releases](https://github.com/CANopenNode/CANopenEditor/releases/tag/v4.2.2).
I'm trying to run EDSEditor.exe from `net6.0-windows` folder.

[image]

```bash
# my command 1
dotnet EDSEditor.exe  --infile something.xdd --outfile something.eds 

# my command 2
dotnet exec EDSEditor.exe  --infile something.xdd --outfile something.eds

# the result for both cases
It was not possible to find any compatible framework version
The framework 'Microsoft.WindowsDesktop.App', version '6.0.0' (arm64) was not found.
  - No frameworks were found.

You can resolve the problem by installing the specified framework and/or SDK.

The specified framework can be found at:
  - https://aka.ms/dotnet-core-applaunch?framework=Microsoft.WindowsDesktop.App&framework_version=6.0.0&arch=arm64&rid=osx.11.3-arm64

 ```
 
 May be have I executed command not correctly?

nimrof, 2024-10-24:
> @nimrof I'm not good on .net. Could you help me with correct command to run?

Yes

```shell
#cd EDSSharp
#dotnet --framwork net6.0 run
```

If that works it should be building the cli program and running it.
With zero mac experience I am not sure how to simply execution after it have been build.

ruziev-dev, 2024-10-24:
@nimrof
If you mean run command in source code `EDSSharp` folder
There is no result 😔

```bash
# command in  EDSSharp folder
dotnet --framwork net6.0 run

# result
Could not execute because the specified command or file was not found.
Possible reasons for this include:
  * You misspelled a built-in dotnet command.
  * You intended to execute a .NET program, but dotnet---framwork does not exist.
  * You intended to run a global tool, but a dotnet-prefixed executable with this name could not be found on the PATH.
 ```

If the framework allows to be executed on MacOS could you post releases with such package in future?

trojanobelix, 2024-11-18:
@nimrof Still in progress, or should we close it?

nimrof, 2024-11-18:
> @nimrof Still in progress, or should we close it?

technical still in progress, but only waiting for a release

nimrof, 2024-11-29:
@ruziev-dev 
https://github.com/CANopenNode/CANopenEditor/releases/tag/v4.2.3

The CLI tool is in the net8.0 folder

ruziev-dev, 2024-11-29:
@nimrof Could you help me with correct cli command?

I have tried but it does not working

```sh
dotnet  EDSSharp.exe  --infile something.xdd --outfile something.eds
``` 
[image]

nimrof, 2024-11-29:
> @nimrof Could you help me with correct cli command?
> 
> I have tried but it does not working
> 
> ```shell
> dotnet  EDSSharp.exe  --infile something.xdd --outfile something.eds
> ```

Hi dont have mac so only testet on linux, and i get the same error if i use dotnet, but if i just use ```./EDSSharp.exe``` it works

ruziev-dev, 2024-11-30:
looks like `dotnet  EDSSharp.dll` works but something wrong.

```sh
➜  net8.0 dotnet  EDSSharp.dll                                                
Usage: EDSEditor --infile file.[xdd|eds] --outfile [valid output file] [OPTIONAL] --type [exporter type]
The output file format depends on --outfile extension and --type
If --outfile extension matcher one exporter then --type IS NOT needed
If --outfile extension matcher multiple exporter then --type IS needed
If --outfile has no extension --type IS needed
Exporter types:
  ElectronicDataSheet [.eds]
  DeviceConfigurationFile [.dcf]
  CanOpenNode [.h,.c]
  CanOpenNodeV4 [.h,.c]
  CanOpenXDDv1.0 [.xdd]
  CanOpenNetworkv1.0 [.nxdd]
  CanOpenXDDv1.1 [.xdd]
  CanOpenXDDv1.1stripped [.xdd]
  CanOpenXDCv1.1 [.xdc]
  CanOpenNetworkXDDv1.1 [.nxdd]
  CanOpenNetworkXDCv1.1 [.nxdc]
  CanOpenNodeProtobuf(json) [.json]
  CanOpenNodeProtobuf(binary) [.binpb]
  DocumentationHTML [.html]
  DocumentationMarkup [.md]
  NetworkPDOReport [.md]
```

But if I call with not enough arguments it shows me usage info, but if arguments are enough it executes but nothing changes in directory.

[image]

nimrof, 2024-12-15:
> looks like `dotnet EDSSharp.dll` works but something wrong.
> 
> ```shell
> ➜  net8.0 dotnet  EDSSharp.dll                                                
> Usage: EDSEditor --infile file.[xdd|eds] --outfile [valid output file] [OPTIONAL] --type [exporter type]
> The output file format depends on --outfile extension and --type
> If --outfile extension matcher one exporter then --type IS NOT needed
> If --outfile extension matcher multiple exporter then --type IS needed
> If --outfile has no extension --type IS needed
> Exporter types:
>   ElectronicDataSheet [.eds]
>   DeviceConfigurationFile [.dcf]
>   CanOpenNode [.h,.c]
>   CanOpenNodeV4 [.h,.c]
>   CanOpenXDDv1.0 [.xdd]
>   CanOpenNetworkv1.0 [.nxdd]
>   CanOpenXDDv1.1 [.xdd]
>   CanOpenXDDv1.1stripped [.xdd]
>   CanOpenXDCv1.1 [.xdc]
>   CanOpenNetworkXDDv1.1 [.nxdd]
>   CanOpenNetworkXDCv1.1 [.nxdc]
>   CanOpenNodeProtobuf(json) [.json]
>   CanOpenNodeProtobuf(binary) [.binpb]
>   DocumentationHTML [.html]
>   DocumentationMarkup [.md]
>   NetworkPDOReport [.md]
> ```
> 
> But if I call with not enough arguments it shows me usage info, but if arguments are enough it executes but nothing changes in directory.
> <img alt="image" width="569" src="https://private-user-images.githubusercontent.com/74172766/391223386-d31ef857-af41-43c3-9560-1a2b9d294135.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyNzk4MjAsIm5iZiI6MTczNDI3OTUyMCwicGF0aCI6Ii83NDE3Mjc2Ni8zOTEyMjMzODYtZDMxZWY4NTctYWY0MS00M2MzLTk1NjAtMWEyYjlkMjk0MTM1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE1VDE2MTg0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlZmIyMmZhZWI3NmFlYjhjYzhlN2Y2NWEwMTBkZjcyN2FmNjlkYjVhM2YzNTcxZTI5NjkwZTRiYTJjYmNhZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0BVcfJdRaybf5LMkk97i6pXilynZILtbAHiR21SV4i0">

Sorry for the delay, it should probably output a error when given a unsupported filetype instead of silence.
xpd is not yet supported on the CLI, xdd and eds only.

Its on the todo-list
https://github.com/CANopenNode/CANopenEditor/issues/122

RomainPelletant, 2025-04-15:
@ruziev-dev may I suggest to use docker & XQuartz as X11 client (repo [here](https://github.com/RomainPelletant/docker_canopeneditor?tab=readme-ov-file)) : GUI and CLI are working properly.