CANopenNode/CANopenEditor #145: Support .Net8 as LTS?

trojanobelix, 2024-11-09:
I have created a net8 Branch for testing. Because it is the next LTS version it might make sense to support it. Seems not to be a big think. @nimrof  @CANopenNode 
Should we or not?

nimrof, 2024-11-09:
Good initiative.
Yes we should.
.net6 has end-of-life in a few days (12. November) and .net8 is the next logical step.
So maybe we should just replace .net6 with .net8?

trojanobelix, 2024-11-09:
Here are the few rough edges I found during the conversion:

- I didn't manage to add 8 to the 4.8 and 6 project. But I didn't bother because I just wanted to see what was needed for 8. 
- The sourcegrid is again maintained by Siemens (https://github.com/siemens/sourcegrid) and is 8. compatible. However, earlier .net versions are no longer supported.
- The Siemens sourcegrid NuGet packet sources are on Github and I have problems to authenticate myself (TOKEN fits...)
- .8 does not allow the use of names like ‘file’ in ‘public file file’ 

Mainly these are problems with the toolchain, the rest seems to be compatible with a few changes

nimrof, 2024-11-09:
> Here are the few rough edges I found during the conversion:
> 
> - I didn't manage to add 8 to the 4.8 and 6 project. But I didn't bother because I just wanted to see what was needed for 8.

I can fix, if you want :), but it might be easier to start from scratch to avoid unnecessary changes the migration does

> - The sourcegrid is again maintained by Siemens (https://github.com/siemens/sourcegrid) and is 8. compatible. However, earlier .net versions are no longer supported.
> - The Siemens sourcegrid NuGet packet sources are on Github and I have problems to authenticate myself (TOKEN fits...)

Dont think we need to change that, cant we use the same we use for .net 6?

> - .8 does not allow the use of names like ‘file’ in ‘public file file’ 

We can just rename it to File.

> Mainly these are problems with the toolchain, the rest seems to be compatible with a few changes 

👍

trojanobelix, 2024-11-09:
In any case, start from the beginning!  I think you can do it in a few minutes! I'm more of an embedded nerd and use 9 commits to do very simple things in C# :-)

trojanobelix, 2024-11-09:
We do not need to change to Siemens jet. But the old package one is really old, l and I do lieke a maintained source. But no need to hurry, if it is not easy to use different packages with different .net versions.

nimrof, 2024-11-10:
> In any case, start from the beginning! I think you can do it in a few minutes! I'm more of an embedded nerd and use 9 commits to do very simple things in C# :-)

take a look at #146
I mostly do embedded c/c++ at work, but c# is just so much better for web/gui if you do not have to share much code. 

> We do not need to change to Siemens jet. But the old package one is really old, l and I do lieke a maintained source. But no need to hurry, if it is not easy to use different packages with different .net versions.

Agree its old, but hopefully its will hold until the new gui is done🤞

trojanobelix, 2024-11-10:
From my point of view: Merge!

nimrof, 2024-11-10:
> From my point of view: Merge!

Then feel free to approve it :D

PR #146 (2024-11-11): Replace .net6 with .net8
Replaced .net6 with .net8 as .net6 has eol in a few days
resolves #145
closed 2024-11-11 by trojanobelix