CANopenNode/CANopenEditor #155: Add missing fields in File Info and Device Info.

ysx3ehaor, 2024-12-27:
Please consider adding the following fields:

- File Info: FileVersion, FileRevision
- Device Info: RevisionNumber

Some software requires these fields to display information when importing EDS files. Currently, I have to manually edit the EDS file text.

Thank you!

nimrof, 2024-12-28:
> Please consider adding the following fields:
>     * File Info: FileVersion, FileRevision

Will look into it

>     * Device Info: RevisionNumber

Looking at #154 that might be fixed by linking it with the value of identity object subindex 03.
Would that work for you?
Might add a textbox that indicates where the values is coming from and some tooltip

trojanobelix, 2024-12-28:
What build? Can not reproduce. Use latest build!

henri62, 2025-03-21:
I can confirm that "FileVersion" does work in the v4.2.3 version of the editor.

But there is no field to fill-in the FileRevision. Thas number is always 1.
I would suggest to add an additional textbox entry in the TAB "Device Info" / File Info section below the File version.

Also the field [DeviceInfo] -> RevisionNumber is missing in the editor, in the generated EDS its always 0.
I would suggest to add a field it just below the Product ID.

Another issue I saw is that the "File Info" and "Product ID" fields are UNSIGNED8 types and you can now fill in anything in that field. 
During save or export it is not checked if its a non-number (hex numbers should be allowed by the way). The result is a non-compliant EDS file.

The CiA306 says this about the type of fields:
```
FileVersion shall indicate the actual file version (Unsigned8),
FileRevision shall indicate the actual file revision (Unsigned8),
....
VendorNumber shall provide the unique vendor ID according to identity object sub-index 01h (Unsigned32)
ProductNumber shall provide the product code according to identity object sub-index 02h (Unsigned32)
RevisionNumber shall provide the product revision number according to identity object subindex 03h (Unsigned32)
```

It would be very nice if these fields are checked during save or export if its really a valid number.

Regarding the linking of the above fields with the OD entries; to my opinion the Device Info and File Info section sould be autoritatitive and the Object dictionary should follow that.

robincornelius, 2026-10-20:
Merged PR #237 adds editable RevisionNumber and OrderCode fields, commits pending Device Info edits, and tests EDS/DCF save/reopen. This is partial progress on the missing-field workflow; File Info coverage and RevisionNumber XDD/protobuf representation still need a complete review. Keeping this broader request open; see also #226.