CANopenNode/CANopenEditor #172: When I want to modify the data type of a member within a sub index, all data types within the sub index are changed simultaneously.

Letian-stu, 2025-03-03:
**Describe the bug**
When I want to modify the data type of a member within a sub index, all data types within the sub index are changed simultaneously.

**Version used**
v4.2.3-0-gc1071ab

**Expected behavior**
Modifying sub member types will not change simultaneously

**Screenshots**
1.
[image]

2.
[image]

nimrof, 2025-03-03:
Hi @Letian-stu,
Thanks for reporting bugs, but unless i am missing something this is not a bug, but a feature.

In your example the object is a array and array is only allowed to have one datatype (except subindex 0 that must be uint8)
So the editor helps you so you do not have to change the datatype of every subindex.

If you want different datatypes, then the object should be a record

trojanobelix, 2025-03-03:
@nimrof  You missed nothing, it is a feature:

Cia DS301 7.4.3 Object code usage:
ARRAY A multiple data field object where each data field is a simple variable of the **SAME basic data** type e.g. array of UNSIGNED16 etc. Sub-index 0 is of UNSIGNED8 and therefore not part of the ARRAY data.

Letian-stu, 2025-03-03:
> Hi [@Letian-stu](https://github.com/Letian-stu), Thanks for reporting bugs, but unless i am missing something this is not a bug, but a feature.
>
> In your example the object is a array and array is [...]

Sorry, it's my fault. I didn't use RECORD

closed 2025-03-03 by Letian-stu