CANopenNode/CANopenEditor #179: Unknown key "NG_Slave" in the EDS file (CTT:Warning).

henri62, 2025-03-21:
**Describe the bug**
I saw that during export of an XDD file a key 'NG_Slave' is written in the [DeviceInfo] section.

The CANopen compliancy test tool (CTT) gives this warning:
warning 21: Unknown or not used entry "NG_Slave" in section [DeviceInfo]

I cannot find any reference in the CiA documents describing this field.

If it is a known parameter, where is it documented? Is it a problem in the CTT or de Edseditor?

**To Reproduce**
Just export any XDD file to EDS format file.

**Version used**
v4.2.3-0-gc1071ab

**Expected behavior**
The generated EDS file should pass the CTT.

**Additional context**
It is related to the Node guarding Slave / Master settings.

nimrof, 2025-03-21:
> **Describe the bug** I saw that during export of an XDD file a key 'NG_Slave' is written in the [DeviceInfo] section.
>
> The CANopen compliancy test tool (CTT) gives this warning: warning 21: Unknown [...]

@trojanobelix looks like you added in the commit below, do you know more about it?
```
[EdsExport]
public bool NG_Slave;
```
https://github.com/CANopenNode/CANopenEditor/commit/ad5da4c43001ff84901e9dbcc622350e34fef04d#diff-113b0e03b0a1c73f021ecddcb275fde0a92ef4ac2c2d914f4c3c2f89c968785fR799-R800 

> Another warning is present with the section/key [FileInfo] / LastEDS but this key is specified in the CiA306 by the CTT complains with this warning:
> Test Warning : .eds(5) : warning 21: Unknown or not used entry "LastEDS" in section [FileInfo].
> I think this is a bug in the CTT. But when its not filled in, cant we just leave out this keyword when its empty?

Sounds good

robincornelius, 2026-10-20:
The LastEDS portion is fixed by merged PR #237: DCF-only fields are explicitly excluded from EDS output, with Issue190LastEdsOnlyAppearsInDcf regression coverage. This does not resolve the NG_Slave specification/CTT question, so this issue remains open.